OmeSatoFoundation/ome2023 #10: growpartコマンドが見つからない

pshimizu, 2022-11-15:
Dockerfileにcloud-guest-utilsのapt installを追加して対策可能。
------------------------------- error --------------------------------
+ IMG_NAME=2022-09-22-raspios-bullseye-arm64.img
+ '[' '!' -e 2022-09-22-raspios-bullseye-arm64.img ']'
+ truncate -s 7800000000 2022-09-22-raspios-bullseye-arm64.img
++ losetup -P -f --show 2022-09-22-raspios-bullseye-arm64.img
+ DEVICE_PATH=/dev/loop0
+ set +e
+ growpart /dev/loop0 2
./contrib/scripts/install.bash: line 50: growpart: command not found
+ EXITCODE=127
+ '[' 127 -ne 0 ']'
+ '[' 127 -ne 1 ']'
+ echo 'growpart unexpectedly exited.'
growpart unexpectedly exited.
+ exit 127
------------------------------ ここまで ---------------------------

diff --git a/Dockerfile b/Dockerfile
index c8b3f33..68d1857 100644
--- a/Dockerfile
+++ b/Dockerfile
@@ -29,6 +29,7 @@ RUN dpkg --add-architecture arm64 \
     && sed -i 's/^deb/deb [arch=amd64,arm64]/g' /etc/apt/sources.list \
     && apt update \
     && apt install -y \
+    cloud-guest-utils \
     dpkg-cross \
     g++-aarch64-linux-gnu \
     libasound2-dev:arm64 \
diff --git a/contrib/Julius b/contrib/Julius
--- a/contrib/Julius
+++ b/contrib/Julius
@@ -1 +1 @@
-Subproject commit 36de4696f9095cd03f2f60b0fb37b80501bf90e9
+Subproject commit 36de4696f9095cd03f2f60b0fb37b80501bf90e9-dirty

RollMan, 2022-11-17:
`cloud-guest-utils` は #8 ですでに含まれており、13 日に `master` にマージ済です。
お手元に最新のソースコードが適用されているか確認をお願いします。

ip-arch, 2022-11-17:
入れ違いだった？mattermostにアナウンスがあってpullしたと思うけど、#8のブランチは見えないね。
アナウンス以前は、そもそもdockerでビルド出来ていない。
いずれにせよ、clould-guest-utilsが入っているなら、issueをclose します。
<pre>
$ git branch -av
* growpart_add                             65a2eb1 add clould-guest-utils to install growpart
  master                                   2aeb129 Update README.md
  remotes/origin/HEAD                      -> origin/master
  remotes/origin/directory_structure_draft acfac31 remove run
  remotes/origin/growpart_add              65a2eb1 add clould-guest-utils to install growpart
  remotes/origin/install_sh_draft          a6f86d0 add install.sh
  remotes/origin/master                    2aeb129 Update README.md
  remotes/origin/material                  96bbbc2 apply ome2019 lecture material

</pre>

closed 2022-11-17 by ip-arch