OmeSatoFoundation/ome2023 #14: Raspberry Pi OSの初回起動時のWizardを出さずにデフォルトユーザとパスワードを設定する

yutashx, 2023-03-02:
現状Raspberry Pi 400用のRaspberry Pi OSは初回起動時にユーザー名などの設定を要求される（Wizardの画面: https://ome-edu.nshimizu.com/index.php?未来塾2023に向けてFabo%20を%20Raspberry%20Pi%20400%20で動作させる#t45f2caf）。
青梅の教科書のパスの表記とOSのパスが一致している方が望ましいため、OSの配布時にユーザー名などをあらかじめ設定しておくことが望ましい。
そのため、Raspberry Pi OSの初回起動時のWizardを出さずに次の項目を設定するようにOSを作成するスクリプト（https://github.com/OmeSatoFoundation/ome2023/blob/master/contrib/scripts/install.bash）を変更したい。

- ユーザー名
- パスワード
- ホームディレクトリ
- 言語
- 国
- タイムゾーン
- キーボード配列

ちなみに`/etc/xdg/autostart/piwiz.desktop`を消去してOSを作成すれば、Wizardを出さずにOSを初回起動できそう。（参考: https://forums.raspberrypi.com/viewtopic.php?t=231557）

RollMan, 2023-03-13:
`/etc/xdg/autostart/piwiz.desktop` は Raspbian 時代のもので、Raspberry Pi OS で初回起動時に現れるものとは異なる気がします。おそらく、起動時に `/sbin/init` を置換するカーネルパラメータで指定されているものとかが該当のウィザードです。

RollMan, 2023-03-14:
たぶんできた　皆さんテストお願いします

- https://adaptive.u-aizu.ac.jp/itschool-raspbian-2023-03-14T09.43.44.img.xz.gpg
- https://adaptive.u-aizu.ac.jp/itschool-raspbian-2023-03-14T09.43.44.img.xz.md5

```bash
gpg --decrypt --output itschool-raspbian-2023-03-14T09.43.44.img.xz itschool-raspbian-2023-03-14T09.43.44.img.xz.gpg
```

または windows なら [kleopatra](https://gpg4win.org/download.html) が正式な gpg4win のインターフェースです

RollMan, 2023-03-14:
pi にしたということは `ome`/ を `/home/pi/ome` に配置すべきなのを忘れていました

RollMan, 2023-03-14:
`/usr/local/share/ome` のコンテンツがない

hogenimushi, 2023-03-15:
本件、以前のディスとリビューションの挙動に合わせるか、新しい挙動に合わせるか、どちらが望ましいか、該当する章の担当者とディスカッションしてはどうでしょうか。

PR #29 (2023-05-16): Restore Where Directories of `01/`, ... are Placed: `/usr/local/share`
Closes #19, closes #14, closes #21

Revert "put lecture materials (01/, ...) into home directory instead of datadir"

This reverts commit 6549d4c7b98d566baee0029df50d7e223bfb89e4. restore
where directories of 01/, ... are placed: /usr/local/share.

Accordingly, I had `make install` run outside `chroot` (#19).
closed 2023-05-16 by RollMan