OmeSatoFoundation/ome2023 #17: Raspberry Pi 400で実行できない可能性のあるHSPファイル

yutashx, 2023-03-14:
対象ファイル
`02/drop.hsp`

実行環境
[30a7f8a](https://github.com/OmeSatoFoundation/ome2023/commit/30a7f8a5f205db806c77ad07a20d863fbfc77d18)でDockerを用いてOSイメージを作成

実行時のログ
```
/home/pi/ome2023/02 ./drop.hsp
hsed: RUN(./hspcmp -d -i -u /home/pi/ome2023/02/__hsptmp.hsp)
#HSP script preprocessor ver3.6 / onion software 1997-2021(c)
#Use file [hspdef.as]
#Use file [dish_enhance.as]
#Use file [dish_enhance.as]
#HSP code generator ver3.6 / onion software 1997-2021(c)
#use UTF-8 strings.
#Uninitalized variable (score).
#Uninitalized variable (bg).
#Uninitalized variable (key).
#Uninitalized variable (id).
#Uninitalized variable (my).
#Uninitalized variable (ax).
#Uninitalized variable (ay).
#Uninitalized variable (res).
#Uninitalized variable (u1).
#Uninitalized variable (u2).
#Uninitalized variable (u3).
#Uninitalized variable (tcol).
#Uninitalized variable (tx).
#Uninitalized variable (ty).
#Uninitalized variable (var_gm).
#Uninitalized variable (var_rate).
#Uninitalized variable (v_gmode).
#Uninitalized variable (v_rate).
#Code size (2096) String data size (457) param size (0)
#Vars (38) Labels (13) Modules (0) Libs (0) Plugins (3)
#No error detected. (total 4085 bytes)

Runtime[hsp3dish].
hsed: Runtime [hsp3dish].
Init:HGIOScreen(640,480)
[MMMan] Failed se_puyo.wav on bank #0 (1)
Execute from __hsptmp.ax runtime[hsp3dish](1).
hsed: Process end 255.

```

RollMan, 2023-03-15:
このコメントはメモですが，

> [MMMan] Failed se_puyo.wav on bank #0 (1)

とあるので，[se_puyo.wav](https://github.com/OmeSatoFoundation/ome2023/blob/30a7f8a5f205db806c77ad07a20d863fbfc77d18/02/se_puyo.wav) が配置されていない可能性がありますね．このコミットでは `/home/pi/ome` のコピーをユーザに各々でしてもらうことになっていたので，原因は二通り考えられます:

* [install.bash](https://github.com/OmeSatoFoundation/ome2023/blob/30a7f8a5f205db806c77ad07a20d863fbfc77d18/contrib/scripts/install.bash) がコピー元 (`/usr/local/share/ome/02/se_puyo.wav`) の配置に失敗している
* ユーザにコピーしてもらうときに失敗している

コピーの失敗にも，ファイルが配置されなかったことだけではなく，読み込み権限が無いなども考えられます．

@yutashx もしすぐできるようであれば，`/usr/pi/ome/02/se_puyo.wav` が存在するか，stat はどうか，wav の meta field の破損はないか，その他，もし存在しなければ，`/usr/local/share/ome/02/se_puyo.wav` が存在するかみてもらえませんか？

yutashx, 2023-03-19:
[itschool-raspbian-2023-03-18T02.04.51.img.gpg](https://adaptive.u-aizu.ac.jp/itschool-raspbian-2023-03-18T02.04.51.img.gpg)のOS上で`ome/02`ディレクトリ下のファイルサイズを確認してみました。
wavファイルが想定されるファイルサイズよりかなり小さいように見えます。

```
$ ls -lh
pi@raspberrypi:~/ome/02 $ ls -lh    
合計 2.1M    
-rw-r--r-- 1 pi pi 2.7K  3月 18  2023 80font.png    
-rw-r--r-- 1 pi pi 4.0K  9月 22 13:50 __hsptmp.ax    
-rw-r--r-- 1 pi pi 3.3K  9月 22 13:50 __hsptmp.hsp    
-rw-r--r-- 1 pi pi 1.3M  3月 18  2023 bg.png    
-rw-r--r-- 1 pi pi 2.7K  3月 18  2023 block3.hsp    
-rw-r--r-- 1 pi pi  132  3月 18  2023 cappi08.ogg    
-rw-r--r-- 1 pi pi  18K  3月 18  2023 chr.png    
-rw-r--r-- 1 pi pi  18K  3月 18  2023 chrorg.png    
-rw-r--r-- 1 pi pi 3.3K  3月 18  2023 drop.hsp    
-rw-r--r-- 1 pi pi  129  3月 18  2023 explos.wav    
-rw-r--r-- 1 pi pi 4.0K  3月 18  2023 fuujin.hsp    
-rw-r--r-- 1 pi pi  14K  3月 18  2023 fuujin.png    
-rw-r--r-- 1 pi pi  113  3月 18  2023 gpio.hsp    
-rw-r--r-- 1 pi pi  161  3月 18  2023 handan.hsp    
-rw-r--r-- 1 pi pi  173  3月 18  2023 hata.hsp    
-rw-r--r-- 1 pi pi   46  3月 18  2023 hsp3dish.ini    
-rw-r--r-- 1 pi pi  129  3月 18  2023 inject.wav    
-rw-r--r-- 1 pi pi 113K  3月 18  2023 jp88bg.png    
-rw-r--r-- 1 pi pi 1.5K  3月 18  2023 jp88chr.png    
-rw-r--r-- 1 pi pi  130  3月 18  2023 jp_dead.wav    
-rw-r--r-- 1 pi pi  129  3月 18  2023 jp_jump.wav    
-rw-r--r-- 1 pi pi  937  3月 18  2023 kakiget.hsp    
-rw-r--r-- 1 pi pi  107  3月 18  2023 keisan.hsp    
-rw-r--r-- 1 pi pi 7.1K  3月 18  2023 koma.png    
-rw-r--r-- 1 pi pi 1.9K  3月 18  2023 landing.hsp    
-rw-r--r-- 1 pi pi  108  3月 18  2023 mes.hsp    
-rw-r--r-- 1 pi pi  145  3月 18  2023 move.hsp    
drwxr-xr-x 2 pi pi 4.0K  3月 18  2023 org    
-rw-r--r-- 1 pi pi 4.2K  3月 18  2023 poker.hsp    
-rw-r--r-- 1 pi pi  140  3月 18  2023 repeat.hsp    
-rw-r--r-- 1 pi pi  131  3月 18  2023 se_block2.wav    
-rw-r--r-- 1 pi pi  130  3月 18  2023 se_break3.wav    
-rw-r--r-- 1 pi pi  129  3月 18  2023 se_foot.wav    
-rw-r--r-- 1 pi pi  129  3月 18  2023 se_jump2.wav    
-rw-r--r-- 1 pi pi  129  3月 18  2023 se_puyo.wav    
-rw-r--r-- 1 pi pi  129  3月 18  2023 sel.wav    
-rw-r--r-- 1 pi pi 4.7K  3月 18  2023 shoot.hsp    
-rw-r--r-- 1 pi pi  699  3月 18  2023 star_liner.hsp    
-rw-r--r-- 1 pi pi  112  3月 18  2023 stars.hsp    
-rw-r--r-- 1 pi pi  318  3月 18  2023 swhandan.hsp    
-rw-r--r-- 1 pi pi  177  3月 18  2023 switch.hsp    
-rw-r--r-- 1 pi pi 3.4K  3月 18  2023 swjump.hsp    
-rw-r--r-- 1 pi pi 497K  3月 18  2023 title.png    
-rw-r--r-- 1 pi pi 4.9K  3月 18  2023 tof2.hsp    
-rw-r--r-- 1 pi pi  13K  3月 18  2023 trump.png    
-rw-r--r-- 1 pi pi  33K  3月 18  2023 uiparts.png 
```

また`se_puyo.wav`の中身をテキストエディタで開くと次のような記述がありました。
```
version https://git-lfs.github.com/spec/v1    
oid sha256:d857ab18e83c3af630fa84b6cb232cbcf3ebd3c00c4c307a421dab23624f6676    
size 2210    
```

RollMan, 2023-03-19:
大きいバイナリデータは GitHub LFS に保存していたのですが，これを fetch するのに別なコマンドが必要なところ，`install.bash` に書き忘れていました．

`contrib/scripts/install.bash` に LFS のデータを fetch させます

closed 2023-03-23 by RollMan