OmeSatoFoundation/ome2023 #72: 第7回 tv.pyがイメージに含まれていない、実行権限が付与されていない

twinkletia, 2023-07-05:
2023-06-17のイメージには、第7回演習tv.hspで使用するtv.pyが含まれていない。
[contrib/HTMLParse/Makefile.am](https://github.com/OmeSatoFoundation/ome2023/tree/master/contrib/HTMLParse/Makefile.am)に含まれていないのが原因？

また、contrib/HTMLParse/tv.pyのパーミッションが622となっているので、pythonスクリプトとして実行できる状態にない。

https://github.com/OmeSatoFoundation/ome2023/blob/307617ccf08f4ddb428099fd91642ad0eaa978af/contrib/HTMLParse/Makefile.am#L1

RollMan, 2023-07-07:
[contrib/HTMLParse/Makefile.am](https://github.com/OmeSatoFoundation/ome2023/tree/master/contrib/HTMLParse/Makefile.am) に `tv.hsp` を含めていないのは，おそらく忘れていました :pray:

RollMan, 2023-07-07:
ついでに，weblio.py の python 3 実装が見当たらず，python2 の実装のままイメージに含めてしまいました．．これはもしかして，HTML の構造が変わってもとのスクリプトではスクレイピングできなくなったなどありましたか？

そういうわけで `weblio.py` は削除予定等変更点ありましたら連絡か PR をお願いします．

twinkletia, 2023-07-07:
はい、仰る通り、weblioのHTMLの構造が以前と変わったので、変更する予定です。後にPRいたします。

PR #112 (2024-06-05): restore tv.py which was accidentially removed at #86
closes #83, closes #72, closes https://github.com/OmeSatoFoundation/ome-doc/issues/449
commit 7fe5dd45 (2024-06-05): restore tv.py which was accidentially removed at #86

closes #83, closes #72, closes https://github.com/OmeSatoFoundation/ome-doc/issues/449
closed 2024-06-05 by RollMan