OmeSatoFoundation/ome2023 #92: dockerコンテナ内でのHSPのmakeに失敗する

twinkletia, 2023-11-16:
READMEで記載されている方法でdocker runすると、HSPをmakeする際にリンクで大量のundefined referenceが出てイメージのビルドができません。

**コマンド**
`git clone git@github.com:OmeSatoFoundation/ome2023.git --recurse && pushd ome2023 && eval $(ssh-agent -s) && ssh-add ~/.ssh/id_rsa && docker run --rm -ti -v /dev/:/dev --privileged -v $(pwd):/work -v --workfdir=/work -v $SSH_AUTH_SOCK:/ssh-agent -e SSH_AUTH_SOCK=/ssh-agent ome2023 sh -c 'aclocal -I m4 && automake -a -c && autoconf && ./configure --build=x86_64-linux-gnu --host=aarch64-linux-gnu --prefix=/usr/local && make -j$(nproc) && ./contrib/scripts/install.bash -f' 2>&1 | tee build.log && popd`

**ログ**
[build.log](https://github.com/OmeSatoFoundation/ome2023/files/13374072/build.log)

RollMan, 2023-11-16:
多分リンカが先走っているような感じです。

まず `make clean` して、

```bash
docker run --rm -ti -v /dev/:/dev --privileged -v $(pwd):/work -v --workfdir=/work -v $SSH_AUTH_SOCK:/ssh-agent -e SSH_AUTH_SOCK=/ssh-agent ome2023 make clean # (動作未確認)
```

`make` の `-j` のオプションを減らして試していただけますか？いちおう、こちらでは `-j 4` で 90% 位の確率でビルドが成功することを確認していますが、100% ではないです。どうしてもうまく行かないときは `-j 1` でやってみてください。
それでも動かない場合は、`-j 1` のときの出漁を添付いただけますか？

#### メモ
たぶん build.log の 1773 行目でエラー

```
aarch64-linux-gnu-g++ -Wno-write-strings --exec-charset=UTF-8 -DHSPDISH -DHSPLINUX -DHSPDEBUG -DUSE_OBAQ src/hsp3/dpmread.do src/hsp3dish/geometry.do src/hsp3/hsp3.do src/hsp3/hsp3code.do src/hsp3/hsp3debug.do src/hsp3dish/hsp3gr_dish.do src/hsp3/hsp3int.do src/hsp3/hspvar_core.do src/hsp3/hspvar_double.do src/hsp3/hspvar_int.do src/hsp3/hspvar_label.do src/hsp3/hspvar_str.do src/hsp3/hspvar_struct.do src/hsp3dish/hspwnd_dish.do src/hsp3dish/hspwnd_obj.do src/hsp3dish/hspwnd_edit.do src/hsp3dish/random.do src/hsp3/stack.do src/hsp3/strbuf.do src/hsp3/strnote.do src/hsp3/linux/hsp3ext_sock.do src/hsp3/linux/hsp3ext_linux.do src/hsp3/linux/devctrl_io.do src/hsp3dish/essprite.do src/hsp3dish/texmes.do src/hsp3dish/sysreq.do src/hsp3dish/emscripten/hgtex.do src/hsp3dish/emscripten/hgiox.do src/hsp3dish/emscripten/mmman.do src/hsp3dish/emscripten/main.do src/hsp3dish/emscripten/stb_image.do src/hsp3dish/emscripten/fontsystem.do src/hsp3dish/obaq/omkedraw.do src/hsp3dish/obaq/hsp3dw.do src/hsp3dish/obaq/game.do src/obaq/physics/rock.do src/obaq/physics/vessel.do src/hsp3dish/linux/hsp3dish.do src/hsp3dish/linux/webtask_linux.do src/hsp3dish/linux/supio_linux.do -s -o hsp3dish -lm -lGL -lEGL -lSDL2 -lSDL2_image -lSDL2_mixer -lSDL2_ttf -lstdc++ -lcurl
```

RollMan, 2023-11-16:
あとは、debian 系なら、ホストで apt install qemu-user-static ができているかどうかご確認ください。

twinkletia, 2023-11-16:
申し訳ありません、qemu-user-staticが入っていませんでした。お騒がせしました。

closed 2023-11-16 by twinkletia