OmeSatoFoundation/ome2023 #99: bookworm のベースイメージを使うと HSP で gpio コマンドが動作しない

RollMan, 2024-05-06:
Inherited from https://github.com/OmeSatoFoundation/ome-doc/issues/450, related to https://github.com/OmeSatoFoundation/ome2023/pull/96

pseudo filesystem を使った gpio へのアクセスは deprecated で 2020 から削除していく事になっていた (https://www.kernel.org/doc/Documentation/gpio/sysfs.txt https://www.kernel.org/doc/Documentation/ABI/obsolete/sysfs-gpio) が、2024-03-12-raspios-bookworm (Linux 5.15.61-v7+) までに削除されたようで (または [不具合かもしれない](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=3d390ce29859e727e7d94786358c193f8f4cd0a9))、このベースイメージを使用すると HSP の gpio コマンドが動作しない。HSP は gpio 制御のために [pseudo filesystem](https://github.com/OmeSatoFoundation/OpenHSP/blob/6888481cf0bc8e1fcacc3b1c6731c16e280c83a7/src/hsp3/linux/devctrl_io.cpp#L403) を使用している。

かわりに、c library として新しく導入された gpio interface である [libgpiod](https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/) を使う必要がある。

- https://lkml.iu.edu/hypermail/linux/kernel/1607.3/00311.html
- https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/gpio/gpiolib.c?h=v5.16&id=3c702e9987e261042a07e43460a8148be254412e

上記に取り組む前に考察の確証を持つため下記を確認する。

- 問題が再現する手順を確立する
- `/sys/class/gpio` の存在・可用性確認
- kernel version の確認

ベースディストリビューションのアップグレード (https://github.com/OmeSatoFoundation/ome2023/pull/96) に関する問題のマイルストーンはその最終的な量によって決める。問題の量が過多の場合は修正を次年度に持ち越し、当該年度は前年度のイメージを再利用する。

RollMan, 2025-05-28:
https://github.com/OmeSatoFoundation/ome2023/pull/96/commits/6608c9434251a269d4c6e58014a0c401425f3cab closes this.

aizuknight, 2025-06-17:
おにたま先生のOpenHSPの以下ようにコードを変更して`make`し、作られたバイナリで`Ltika.hsp`を実行したらLEDが点滅した。
https://github.com/onitama/OpenHSP/blob/9553278dde6659bc980721729971bde0e143bffa/src/hsp3/linux/devctrl_io.cpp#L551
を
```
gchip = gpiod_chip_open_lookup("/dev/gpiochip4");
```
に変更した

RollMan, 2025-06-17:
User IO assignment has been gotten back into gpiochip0 in the latest bookworm. https://forums.raspberrypi.com/viewtopic.php?t=376470

RollMan, 2025-06-17:
NOTE: `gpio_chip_opne_lookup()` is deprecated.

aizuknight, 2025-07-16:
ベースイメージを`2025-05-13-raspios-bookworm-arm64.img`に変更した [94e7a6b](https://github.com/OmeSatoFoundation/ome2023/tree/94e7a6b8553c25b3890a433a99ffb76979949a86) でビルドしたOSにおいて:
- raspi-configでSPIとI2Cが有効になっていることを確認しました.
- Ltika.hspでGPIOが動作することを確認しました.
- anain.hspでSPIが動作することは確認できませんでした. (以前と変わらず-1が表示され続ける)
- oled.hspでI2Cが動作することを確認しました.
- デスクトップ背景がfishermanになってしまっていることを確認しました.
- 日本語入力ができないことを確認しました.

RollMan, 2025-07-16:
ベースイメージを更新した版を https://github.com/OmeSatoFoundation/ome2023/pull/96 に取り込みましょう。

RollMan, 2025-07-27:
gpio のバンク番号に影響を及ぼすパッケージを特定する。

RollMan, 2025-08-13:
[2026v0.1](https://github.com/OmeSatoFoundation/ome2023/releases/tag/2026v0.1) (ベース: `2024-03-12-raspios-bookworm-arm64.img`) に対して，`raspi-firmware` (依存パッケージなし) または `linux-image-*` (と依存の `linux-{image,header}-*`) を更新すると user-facing gpio が `/dev/gpiochip0` にアサインされる．

カーネルリポジトリに該当しそうなコミットを見つけた．https://togithub.com/raspberrypi/linux/pull/6144